ros-industrial/motoman #407: B and T joint - offset or no offset

Doomerdinger, 2021-05-20:
I notice that in some of the robot models (such as the GP180_120) the T joint is offset from the B joint (as it seems to be specified in the robot spec) but in other models (such as the ma2010) the T link transform and the B link transform are coincident. 

Should one of these be preferred when making models or is it an arbitrary choice?

gavanderhoorn, 2021-05-23:
It's not really arbitrary, but there is some history.

Most (all?) robot support packages in ROS-Industrial (ie: those for other OEMs) have the last link offset from the previous-to-last. That approach was also used here for many of the models.

Lately, @EricMarcil commented that it may actually be more conform the way the Yaskawa controller models things to do it the other way around.

This would be more aligned with the D-H parameters (IIRC).

From the context in which you are asking this question (or: the one I assume), I expect you to have a preference for following the D-H parameters more closely.

An appeal to tradition is of course not necessarily a good argument, but if we'd like to align with the other repositories, we'd do it the other way.

Doomerdinger, 2021-05-24:
Internally we actually use the offset. It does not make a huge difference to me one way or the other so long as it is consistent. 

If we generally have the offset, I will keep it that way in my models so it is consistent, then if/when we decide to remove that offset it can be done all at once. Any qualms with that approach?

EricMarcil, 2021-05-25:
On the model I made, I ususally put the offset <joint name="${prefix}joint_6_t-tool0" type="fixed"> like the ma2010.
It's mainly because that's the way it is the MotoSim (Motoman robot simulator) and I usually use those models as reference to make the URDF.  
But I don't really see a problem in doing it the other way.  It might be better to review the models to make it consistent with what is most common for ROS users...