ros-industrial/motoman #408: Is it safe to use GP 180 support for MH 180?

tataetae, 2021-05-23:
We have a MH180 robot, but there is only support for the GP180 for ros-industrial at the moment. Since the GP 180 has exact same joint kinematic (not torque, speed, etc) spec. Would it be safe to use the GP 180 support for MH 180? Or would it be easier to create a PR for MH180 support based upon the GP 180 model? Thanks!

gavanderhoorn, 2021-05-23:
The issue is not really the kinematics -- which, as you mention, are pretty much the same between the models: support packages contain meshes, and those meshes define the *shape* of the robot. That shape is used by tools like MoveIt and Tesseract to perform collision avoidant motion planning.

If the shape is incorrect, you run the risk of motion planning returning paths which could result in collisions.

> would it be easier to create a PR for MH180 support based upon the GP 180 model? 

I'd suggest waiting on a response by @EricMarcil to see whether the two models are sufficiently similar or not.

Creating a support package is certainly not that difficult, especially not when only the names and meshes are different.

So if that would be needed, it shoudn't take too long (but someone will have to do it of course).

EricMarcil, 2021-05-25:
The MH180 model is available under the motoman experimental (not sure why it is there... maybe it wasn't reviewed?).
https://github.com/ros-industrial/motoman_experimental
The package is called motoman_mh_support.

EricMarcil, 2021-05-25:
Actually, the package is for the MH180-120 but the meshes for the MH180 are there.  
We just need to create the urdf file to go with it.

To answer the original question, the GP180 also contains only the GP180-120 variant of the model.  So you can't use that for MH180.  There are actually a few variants of that model.  Can you confirm you're exact robot model.  There should be a tag on the back of the robot or in the pendant you can look under SYSTEM INFO --> VERSION.  The basic MH180 should be MH180-A00

EricMarcil, 2021-05-25:
There is Doomerdigger that currently has a PR #395 to add the real GP180 to the existing GP180-Support.  You can probably use that model.  The only difference between the MH180 and GP180 is that the "Effort" values (Acc/Dec Max Torque) are little higher for the GP180 but I don't think that is generally used.

EricMarcil, 2021-06-04:
@tataetae Where you able to get your model to work?  Or are you still looking for help on this?