ros-industrial/motoman #409: Good idea to use ros1bridge with Moveit2 and ROS2 rolling?

tataetae, 2021-05-27:
To my understanding, this repository is for ROS 1 . So if we are looking to use ROS 2 (as it's probably more future proof as we are in early stage of our project), would it be a good idea to use this repository in conjunction with ROS1 bridge? Our task is mainly pick and place and palletizing/depalletizing (with probably some trimming/sanding/soft object manipulation in the future). Would be glad to hear more about the effort for ROS 2 on motoman robots! Thanks!

gavanderhoorn, 2021-06-01:
It's been a while since I last tried this, but in principle it's possible.

Whether it would be a good idea is a different question.

The driver's default interface is *trajectory* based. So clients send trajectory *goals* and then wait for completion of trajectory execution. That's really low volume traffic, which the bridge should be able to handle.

You would need something like https://github.com/ros2/ros1_bridge/pull/256 though, as it's all [actionlib](https://wiki.ros.org/actionlib) based, which `ros1_bridge` doesn't support by default.

All the high volume traffic will then stay "behind" the bridge, in "ROS 1 land".

If you'd also need `JointState` information on the ROS 2 side, that might change things.

Technically, the bridge should be able to handle it, but I've observed (and heard from other users) different things. Stability might become a problem, but it's not something I can say something conclusive about here. That would have to be tested.

gavanderhoorn, 2021-06-01:
Note btw:

> Our task is mainly pick and place and palletizing/depalletizing (with probably some trimming/sanding/soft object manipulation in the future)

the first two use-cases are rather different than the latter ones.

still possible with a trajectory-based interface, but if "trimming/sanding/soft object manipulation" requires closed-loop control, it's going to be difficult.

Perhaps something like [tingelst/motoman_hardware](https://github.com/tingelst/motoman_hardware) would work better.

But note that it's not compatible with `motoman_driver` immediately.

gavanderhoorn, 2021-08-11:
Closing due to inactivity.

closed 2021-08-11 by gavanderhoorn