ros-industrial/motoman #410: Error 5011 after EStop (ROS_RESULT_NOT_READY_PFL_ACTIVE)

ted-miller, 2021-05-27:
@EricMarcil: Can you look at this issue? I don't understand all the signals you're monitoring in Ros_Controller_IsPflActive.

Steps to reproduce:
- Start a motion trajectory
- EStop during motion
- Release the EStop
- Send Start-Trajectory-Mode
- Check Motion Ready returns 5011

The only way to clear the 5011 is to explicitly send a Stop-Trajectory-Mode message and then try again.
Note that this was performed on a GP-12 that does not have FSU. So... there's definitely nothing going on with PFL.

EricMarcil, 2021-05-27:
@ted-miller: Do you have anything on your system interacting with outputs 15120, 15124 or 15125 on that system?  Those are outputs used by the Avoidance function that affect that logic.
We should probably figure out a way to identify if the robot is a HC robot or not, and put an overall condition around all the PFL related code.  I'll try to look into it tomorrow or early next week.

ted-miller, 2021-05-27:
> Do you have anything on your system interacting with outputs 15120, 15124 or 15125 on that system?

Not that I'm aware of. I can double check later.

> We should probably figure out a way to identify if the robot is a HC robot or not

In UWI, I look at the PFL function parameter. I could bake that into the parameter extraction library.

EricMarcil, 2021-06-02:
I've confirmed the issue.  When the PFL activates, the mpExRcsIncrementMove returns E_EXRCS_IMOV_UNREADY and I had put a flag to mark that state because I had found a mode where other signals couldn't be relied on.  It logs:
    mpExRcsIncrementMove returned UNREADY: %d (Could be PFL Active)
But the E-Stop causes the same state (and maybe the hold, I'll check).
I'll add a check for E-Stop and Hold to distinguish these cases.  I'll also try to add a check to see if the PFL is even active on the system or not.

EricMarcil, 2021-06-04:
The should be corrected by PR #412

gavanderhoorn, 2021-06-11:
#412 should fix this.

closed 2021-06-11 by gavanderhoorn