ros-industrial/motoman #416: Inertial Tags for GP12 and MH12

kumar10725, 2021-07-20:
Inertial and mass tags are missing from the all the urdf models. Without these it is not possible to run the dynamic simulations for the robots, such as estimation of required torque to move an object. Is there any way to obtain those tags? or support will be missing for these tags?

EricMarcil, 2021-07-20:
At this time, Yaskawa doesn't want to publicly publish those values.  To obtain them, you'll need to sign a Non-Discloser Agreement with your local Yaskawa representative.
You can contact me for the America or if you have difficulty figuring out your local representative at: 
eric.marcil@motoman.com

closed 2021-07-21 by kumar10725

kumar10725, 2022-04-20:
@EricMarcil After half a year of discussion with our local representative. They have denied any knowledge of such parameters. I have sent you an email from my work email. We are willing to sign a NDA if necessary.

kumar10725, 2022-04-20:
@EricMarcil  I have sent you an email with the requested details. Thank you very much for your help.

EricMarcil, 2022-04-21:
I've contacted the local Yaskawa representative and they will proceed with establishing the NDA.  Once that is done, I will be able to send the information.

For people that may run into similar situation in the future, some local Yaskawa representative might not be aware of ROS and how to get Kinematic information.  Please direct them to me at:  eric.marcil@motoman.com
I'll be able to explain that we only need them to put in place the NDA and that I will take care of the support afterward.