ros-industrial/motoman #419: Support for ROS_MSG_MOTO_SELECT_TOOL

robertjbush, 2021-08-18:
## Description
When tool changing in ROS, it is also important to change the set tool on the controller (according to Yaskawa Motoman YRC1000micro manual and pendant display messages **not doing this can cause premature failure of the robot**), however it is unclear whether `ROS_MSG_MOTO_SELECT_TOOL` has been exposed as a service or action server. There doesn't seem to be an [existing simple_message](https://github.com/ros-industrial/motoman/tree/kinetic-devel/motoman_driver/src/simple_message) for the functionality.

## My environment
ROS Noetic in a docker container with Ubuntu 20.04
Moveit! version 1.1.5
ros-industrial-client with modified motoman driver v1.9.5 ([our own point streamer branch](https://github.com/additiveautomations/motoman/tree/point-streamer))
motoman gp8 robot

## Problem
There seems to be no existing functionality to use `ROS_MSG_MOTO_SELECT_TOOL` and also it is unclear if the low-level control will actually use it even if it's selected.

## Action Taken So Far
Read manuals and looked through which [simple_messages have been defined already](https://github.com/ros-industrial/motoman/tree/kinetic-devel/motoman_driver/src/simple_message).

## Expected behaviour
Set tool through ROS and see the pendant update the selected tool.

## Actual behaviour
The set tool doesn't change even if a tool is changed through ROS.

EricMarcil, 2021-08-18:
@robertjbush
The SelectTool has been implemented in the MotoRos driver running on the controller.  It was added with the support of HC collaborative robot as we anticipated that it would eventually be needed.  Collaborative robot need more accurate tool definition to operate properly.  But we haven't implemented the ROS API side, since those changes are usually done by people in the ROS community. 
I don't think that anyone has pick-up the task of implementing that part so far.  The expected simple message structure would be:
```
struct _SmBodySelectTool
{
	int groupNo;  				// Robot/group ID;  0 = 1st robot 
	int tool;					// Tool no for the selected group
	int sequence;				// Optional message tracking number that will be echoed back in the response.
} __attribute__((__packed__));
typedef struct _SmBodySelectTool SmBodySelectTool;
```
I'm not sure if you are up for the task.  Otherwise, maybe we can try to reach-up to some other collaborators...

robertjbush, 2021-08-19:
Hey Eric, I spoke to the team and we concluded:

- We're up for it, but not right now because;
- Our software engineers don't have any spare capacity for the next 6 months and;
- Our desire to implement this relates to extending the life of the robot. We are machine builders and need our machines to last 6 years, however we are still a few months away from MVP. But;
- We are recruiting so may be able to bring this forward.

gavanderhoorn, 2021-12-12:
WIP PR: #436.

gavanderhoorn, 2021-12-14:
Just to clarify:

> Set tool through ROS and see the pendant update the selected tool.

this (ie: "see the pendant update the selected tool") will not happen with the current implementation of `SELECT_TOOL`.

See also the discussion in #440 and #436.

closed 2022-02-03 by gavanderhoorn