ros-industrial/motoman #428: unable to set the robot_description parameter

adhitir, 2021-09-28:
Robot: HC10DT
ROS Version: Ubuntu 18:04 with Melodic through Windows WSL

I've been following the tutorials here (http://wiki.ros.org/motoman_driver/Tutorials/Usage) to get a HC10DT robot working with MoveIt! So far, I have had little success.
The biggest issue seems to be with the robot_description parameter. 
The tutorial says to use 
```
rosparam load <path to URDF> robot_description
```

which cannot be used directly as a xacro file is provided. I added these lines in the robot_interface launch file
```
<param name="robot_description"  command="xacro --inorder '$(find motoman_hc10_support)/urdf/hc10dt.xacro'" />
```

and I've used the joint names specified in the motoman_hc10_support/config/joint_names_hc10dt.yaml file.

Firstly, I get this warning when I run the launch file:
```
SUMMARY                                                                                           
========                                                                                                                                                                                            PARAMETERS                                                                                        
* /controller_joint_names: ['joint_1_s', 'jo...                                                   
* /robot_description: <?xml version="1....                                                        
* /robot_ip_address: 192.168.131.40                                                               
* /rosdistro: melodic                                                                             
* /rosversion: 1.14.11                                                                                                                                                                             
NODES                                                                                              
 /                                                                                                   
joint_state (motoman_driver/robot_state)                                                         
joint_trajectory_action (industrial_robot_client/joint_trajectory_action)                         
motion_streaming_interface (motoman_driver/motion_streaming_interface)                                                                                                                          
ROS_MASTER_URI=http://192.168.131.55:11311                                                                                                                                                          
process[joint_state-1]: started with pid [15254]                                                  
process[motion_streaming_interface-2]: started with pid [15255]                                   
process[joint_trajectory_action-3]: started with pid [15256]                                     
[ERROR] [1632863304.989646800]: Failed to find topic_list parameter                               
[ERROR] [1632863305.025623000]: Failed to find topic_list parameter                               
[ WARN] [1632863305.071487400]: Tried to connect when socket already in connected state                                                                                                                             
```

The move_to_joint.py is missing in the current driver by the way. I had an older version of the motoman driver, so I extracted the file from there and used it. I gave the code the current joint state of the robot and I got this error:

```
rosrun motoman_driver move_to_joint.py "[-0.038855452090501785, 0.03427538275718689, 0.0054276310838758945, 0.028037024661898613, -0.010736207477748394, -0.1180443912744522]"                                      
[ERROR] [1632863782.615182]: Start and End position joint_names mismatch                                                
[ERROR] [1632863782.618992]: Unable to move to commanded position. Aborting.                                            Traceback (most recent call last):                                                                                        
File "/home/adhitir/ros_ws/src/nsf-hc10dt/ros-stacks/motoman/motoman_driver/src/move_to_joint.py", line 133, in <module>                                                                                                                          main(sys.argv[1:])                                                                                                    
File "/home/adhitir/ros_ws/src/nsf-hc10dt/ros-stacks/motoman/motoman_driver/src/move_to_joint.py", line 127, in main      move_to_joint(end_pos, duration)                                                                                      
File "/home/adhitir/ros_ws/src/nsf-hc10dt/ros-stacks/motoman/motoman_driver/src/move_to_joint.py", line 69, in move_to_joint                                                                                                                      traj = build_traj(get_cur_pos(), end_pos, duration)                                                                   
File "/home/adhitir/ros_ws/src/nsf-hc10dt/ros-stacks/motoman/motoman_driver/src/move_to_joint.py", line 25, in build_traj                                                                                                                         raise                                                                                                               
TypeError: exceptions must be old-style classes or derived from BaseException, not NoneType
```

When I try and run the moveit package it get the following errors:
```

ROS_MASTER_URI=http://192.168.131.55:11311                                                                                                                                                                                                      
process[robot_state_publisher-1]: started with pid [20194]                                                              
process[move_group-2]: started with pid [20195]                                                                         
process[rviz_AUE_ROBOTLAB_D2_20170_2384831785494813562-3]: started with pid [20196]                                     
[ WARN] [1632864692.601394400]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.                             
[ INFO] [1632864692.696544200]: Loading robot model 'motoman_hc10dt'...                                                 
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-adhitir'                                           
[ INFO] [1632864692.767449200]: rviz version 1.13.18                                                                    
[ INFO] [1632864692.767738600]: compiled against Qt version 5.9.5                                                       
[ INFO] [1632864692.768135900]: compiled against OGRE version 1.9.0 (Ghadamon)                                          
[ INFO] [1632864692.780488000]: Forcing OpenGl version 0.                                                               
[ WARN] [1632864693.062927200]: Kinematics solver doesn't support #attempts anymore, but only a timeout.                
Please remove the parameter '/robot_description_kinematics/hc10dt_arm/kinematics_solver_attempts' from your configuration.                                                                                                                      
[ WARN] [1632864693.079083900]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.                             
[... 73 lines cut ...]
[ INFO] [1632864703.417410900]: hc10dt_arm[RRT]: Created 33 states                                                      
[ INFO] [1632864703.419542900]: hc10dt_arm[RRT]: Created 40 states                                                      
[ INFO] [1632864703.419864200]: ParallelPlan::solve(): Solution found by one or more threads in 0.027931 seconds        
[ INFO] [1632864703.420445900]: hc10dt_arm[RRT]: Starting planning with 1 states already in datastructure               
[ INFO] [1632864703.420706000]: hc10dt_arm[RRT]: Starting planning with 1 states already in datastructure               
[... cut ...]
[ERROR] [1632864705.884026500]: Known controllers and their joints:                                                                                                                                                                             
[ INFO] [1632864705.886098400]: ABORTED: Solution found but controller failed during execution                                                                                                                            
```

What steps am I missing? Why am I not able to control the robot with MoveIt?

gavanderhoorn, 2021-09-29:
Have you seen #415?

adhitir, 2021-10-06:
Yes, I have now. Not sure how it is relevant. Could it be a little more specific.

gavanderhoorn, 2021-10-06:
If your main (first?) goal is to get the driver started, I would recommend to reuse the existing `motoman_hc10_support` package and start:

```
roslaunch motoman_hc10_support robot_interface_streaming_hc10dt.launch robot_ip:=192.168.131.40 controller:=yrc1000
```

That should start everything up as needed for the base driver.

You should not need to create any files yourself for this.

Note that I would not expect this to work with the original version of WSL, and there could be networking issues with WSL2.