ros-industrial/motoman #444: ROS Client and Server differ on what messages should be communicated to each other.  GP7 launch file

rsrisamang-miso, 2021-12-21:
Tried to launch a launch file that defines the topic_list parameter to resolve the missing topic_list error.  Launch file:

<launch>
        <arg name="robot_ip" />

  
  <arg name="controller"/>

        <rosparam command="load" file="$(find motoman_gp7_support)/config/joint_names_gp7.yaml" />
        <rosparam command="load" file="$(find motoman_gp7_support)/config/gp7_motion_interface.yaml" />

        <include file="$(find motoman_driver)/launch/robot_interface_streaming_$(arg controller).launch">
                <arg name="robot_ip"   value="$(arg robot_ip)" />
        </include>
</launch>

where gp7_motion_interface.yaml is defined as:

topic_list:
    - name: gp7_yrc1000_controller
      ns: gp7
      group: 0
      joints: ['joint_1_s', 'joint_2_l', 'joint_3_u', 'joint_4_r', 'joint_5_b', 'joint_6_t']

Received the error from the ROS Server below:

process[motion_streaming_interface-3]: started with pid [225656]
process[io_relay-4]: started with pid [225679]
[joint_state-2] process has died [pid 225651, exit code -5, cmd /root/workspace/devel/lib/motoman_driver/robot_state __name:=joint_state __log:=/root/.ros/log/749845b4-61ef-11ec-b911-0242ac120005/joint_state-2.log].
log file: /root/.ros/log/749845b4-61ef-11ec-b911-0242ac120005/joint_state-2*.log
process[joint_trajectory_action-5]: started with pid [225720]
**[ERROR] [1640044164.085939981 /joint_trajectory_action] [/tmp/binarydeb/ros-kinetic-roscpp-1.12.17/src/libros/publication.cpp:Publication::validateHeader:502]: Client [/motion_streaming_interface] wants topic /joint_path_command to have datatype/md5sum [motoman_msgs/DynamicJointTrajectory/81bfbf2d02070fdef3a528bd72b49257], but our version has [trajectory_msgs/JointTrajectory/65b4f94a94d1ed67169da35a02f33d3f]. Dropping connection.**

Is this a configuration issue or must I make changes in the /motion_streaming_interface node to use trajectory_msgs/JointTrajectory type instead of motoman_msgs/DynamicJointTrajectory?

I checked and I'm running the latest from motoman github, with yrc1000u controller (micro)

gavanderhoorn, 2021-12-21:
Do you actually have a multi-group system?

The `ERROR` isn't really an error in the case of a single group setup.

For your setup specifically: if you don't have a multi-group system, you don't need the `topic_list` parameter. The `.launch` files in the `motoman_gp7_support` package should work for you.

This seems like it was discussed in #235 and #242, which refer to #113 (which discusses the `ERROR` which shouldn't be an `ERROR`).

rsrisamang-miso, 2021-12-21:
No, I don't have a multi-group system, but I could be mistaken. Thank you
for the reply!  I will look at the other similar issues.  I was actually
trying to address another issue in which I was receiving alternating
/joint_state frames with actual joint states in one frame and zero valued
joint states in the following frame.  I did a Wireshark capture and sent to
Ted Miller at Motoman.  He says I have a multi-group system.  My GP 7 has
an external motor attached to the yrc1000u that we will use to slide the
base of the arm along a linear rail.  Does my system constitute a
multi-group system then?

gavanderhoorn, 2021-12-21:
> My GP 7 has an external motor attached to the yrc1000u that we will use to slide the base of the arm along a linear rail. 

depending on how this is setup, that could be a multi-group system, yes. And if @ted-miller says it's multi-group, it most likely is.

If you have one, then I would expect both groups to be mentioned in the `topic_list` configuration you show, but there's only a single group there.

Could you share your complete configuration (ie: including the rail), or perhaps a configuration which is similar to your setup (but not the real packages, if that's sensitive)?

akashjinandra, 2021-12-21:
if it is indeed a multi group system follow this: http://wiki.ros.org/motoman_driver/Tutorials/Creating%20a%20Dual-Arm%20System this has the topic list defined. You also will need to do include the correct file in your ROS job that runs on the controller, in your case, I believe it should be: https://github.com/ros-industrial/motoman/tree/kinetic-devel/motoman_driver/Inform/DX200%2C%20FS100%2C%20YRC1000/single_arm_with_base_axis

ted-miller, 2021-12-21:
It is an R1+S1 system. We have discussed the `Creating a Dual Arm System` link. But, we're investigating another issue first. It seems that the INIT_ROS is not starting with the `robot_enable` service. I have provided my windows-message-emulator utility to validate that the robot is properly configured (eliminate any variables from the ROS configuration).

gavanderhoorn, 2021-12-22:
I'm not entirely sure MotoROS-related problems can cause the error shown.

It would be good -- in parallel to any diagnostics you are performing -- for @rsrisamang-miso to share the exact files (s)he is using (including any customisations), and the exact sequence of commands.

gavanderhoorn, 2021-12-22:
And just to point to something (perhaps) obvious: ROS Kinetic has been EOL for quite some time now. If that version is being used in order to be able to run the packages in this repository: that would not be necessary.

rsrisamang-miso, 2021-12-22:
The current launch file I have is fairly straight forward:
`<launch>
    <arg name="robot_ip" default="10.191.0.40" />
    <rosparam command="load" file="$(find motoman_gp7_support)/config/joint_names_gp7.yaml" />
    <rosparam command="load" file="$(find motoman_gp7_support)/config/gp7_motion_interface.yaml" />

    <include file="$(find motoman_driver)/launch/robot_interface_streaming_yrc1000.launch">
        <arg name="robot_ip" value="$(arg robot_ip)" />
    </include>
</launch>
`

The gp7_motion_interface.yaml file looks like this: 
`topic_list:
    - name: gp7_yrc1000u
      ns: gp7
      group: 0
      joints: ['joint_1_s', 'joint_2_l', 'joint_3_u', 'joint_4_r', 'joint_5_b', 'joint_6_t']
    - name: gp7_yrc1000u
      ns: gp7
      group: 1
      joints: ['joint_1_rail']
`

The joint_names_gp7.yaml file looks like this:
`controller_joint_names: ['joint_1_s', 'joint_2_l', 'joint_3_u', 'joint_4_r', 'joint_5_b', 'joint_6_t']
`
Should I probably add the 'joint_1_rail' joint name at the end of the controller_joint_names list?

I need to run the launch files again in the morning to get the output from using the main launch file described above.

akashjinandra, 2021-12-22:
I think you might want to have a different name for your controller for the second one, like this: 
[image]

You see they have a controller called: sda10f_r1_controller for just the arm and another controller for the base axis: sda10f_b1_controller

rsrisamang-miso, 2021-12-22:
Thanks Akash.  You're right.  I updated my topic_list by changing the name.  That helped clear up the following error messages I was getting:

`[ERROR] [1640192890.751970922 /motion_streaming_interface] [/tmp/binarydeb/ros-kinetic-roscpp-1.12.17/src/libros/service_manager.cpp:ServiceManager::advertiseService:137]: Tried to advertise a service that is already advertised in this node [/gp7/gp7_yrc1000u/stop_motion]
[ERROR] [1640192890.751996204 /motion_streaming_interface] [/tmp/binarydeb/ros-kinetic-roscpp-1.12.17/src/libros/service_manager.cpp:ServiceManager::advertiseService:137]: Tried to advertise a service that is already advertised in this node [/gp7/gp7_yrc1000u/joint_path_command]
`
I no longer receive the messages above when starting the launch file.

I still get this Error message though:
`[ERROR] [1640193800.196059570 /joint_trajectory_action] [/tmp/binarydeb/ros-kinetic-roscpp-1.12.17/src/libros/publication.cpp:Publication::validateHeader:502]: Client [/motion_streaming_interface] wants topic /joint_path_command to have datatype/md5sum [motoman_msgs/DynamicJointTrajectory/81bfbf2d02070fdef3a528bd72b49257], but our version has [trajectory_msgs/JointTrajectory/65b4f94a94d1ed67169da35a02f33d3f]. Dropping connection.
`
Despite the 'Dropping connection' error, I'm able to still receive valid joint_states frames from the rostopic.

gavanderhoorn, 2021-12-22:
@rsrisamang-miso: could you please post the exact sequence of commands and the output you get (as I asked before)?

Make sure to append `--screen` to the `roslaunch` invocation.

Without more information we'll have to keep guessing, like @akashjinandra is doing.

rsrisamang-miso, 2021-12-22:
@gavanderhoorn , apologies for not posting this sooner.  Thank you for also the tips on how to post code using Github code blocks.  Hopefully this is in the correct format:

Filename: miso_yaskawa.launch (motoman_driver/launch)
```
<launch>
    <arg name="robot_ip" default="10.191.0.40" />
    <rosparam command="load" file="$(find motoman_gp7_support)/config/joint_names_gp7.yaml" />
    <rosparam command="load" file="$(find motoman_gp7_support)/config/gp7_motion_interface.yaml" />

    <include file="$(find motoman_driver)/launch/robot_interface_streaming_yrc1000.launch">
        <arg name="robot_ip" value="$(arg robot_ip)" />
    </include>
</launch>
```

Filename : joint_names_gp7.yaml (motoman_gp7_support/config)
```
controller_joint_names: ['joint_1_s', 'joint_2_l', 'joint_3_u', 'joint_4_r', 'joint_5_b', 'joint_6_t']
```

Filename : gp7_motion_interface.yaml (motoman_gp7_support/config)
```
topic_list:
    - name: gp7_yrc1000u
      ns: gp7
      group: 0
      joints: ['joint_1_s', 'joint_2_l', 'joint_3_u', 'joint_4_r', 'joint_5_b', 'joint_6_t']
    - name: gp7_yrc1000u_rail
      ns: gp7
      group: 1
      joints: ['joint_1_rail']
```

Command sent is:
```
roslaunch motoman_driver miso_yaskawa.launch
```

--screen
```
# roslaunch motoman_driver miso_yaskawa.launch 
... logging to /root/.ros/log/282412ee-6361-11ec-8183-0242ac120005/roslaunch-flippy-234719.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://flippy:40685/

SUMMARY
========

PARAMETERS
 * /controller_joint_names: ['joint_1_s', 'jo...
 * /robot_ip_address: 10.191.0.40
 * /rosdistro: kinetic
 * /rosversion: 10.2.0
 * /topic_list: [{'joints': ['joi...

NODES
  /
    io_relay (motoman_driver/io_relay)
    joint_state (motoman_driver/robot_state)
    joint_trajectory_action (industrial_robot_client/joint_trajectory_action)
    motion_streaming_interface (motoman_driver/motion_streaming_interface)

auto-starting new master
process[master]: started with pid [234767]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 282412ee-6361-11ec-8183-0242ac120005
process[rosout-1]: started with pid [234799]
started core service [/rosout]
process[joint_state-2]: started with pid [234810]
process[motion_streaming_interface-3]: started with pid [234817]
process[io_relay-4]: started with pid [234842]
process[joint_trajectory_action-5]: started with pid [234897]
[ERROR] [1640202949.338446697 /joint_trajectory_action] [/tmp/binarydeb/ros-kinetic-roscpp-1.12.17/src/libros/publication.cpp:Publication::validateHeader:502]: Client [/motion_streaming_interface] wants topic /joint_path_command to have datatype/md5sum [motoman_msgs/DynamicJointTrajectory/81bfbf2d02070fdef3a528bd72b49257], but our version has [trajectory_msgs/JointTrajectory/65b4f94a94d1ed67169da35a02f33d3f]. Dropping connection.
```

Please let me know if you'd like any other information.  Is there a better way to set the controller_joint_names?  We have 7 but in the file I just posted, there's only 6 so I'm sure I have something to fix there.  I saw in the other projects the .xacro file was used instead?

I was able to bypass the issues I had earlier by insuring that all the joints on my robot were intialized to Home position.  I was missing the 7th joint and therefore the /robot_enable command was not working.  It now is returning True for success state when sending the /robot_enable command.

rsrisamang-miso, 2021-12-22:
When I send the rosservice list command, I get the following output:

--screen
```
# rosservice list
/gp7/gp7_yrc1000u/joint_path_command
/gp7/gp7_yrc1000u/stop_motion
/gp7/gp7_yrc1000u_rail/joint_path_command
/gp7/gp7_yrc1000u_rail/stop_motion
/io_relay/get_loggers
/io_relay/set_logger_level
/joint_path_command
/joint_state/get_loggers
/joint_state/set_logger_level
/joint_trajectory_action/get_loggers
/joint_trajectory_action/set_logger_level
/motion_streaming_interface/get_loggers
/motion_streaming_interface/set_logger_level
/read_single_io
/robot_disable
/robot_enable
/rosout/get_loggers
/rosout/set_logger_level
/rostopic_235548_1640211125205/get_loggers
/rostopic_235548_1640211125205/set_logger_level
/stop_motion
/write_single_io
```

I tried sending a rosservice call to /gp7/gp7_yrc1000u/joint_path_command and the output I get is:
--screen
```
# rosservice call /gp7/gp7_yrc1000u/joint_path_command "trajectory:
  header:
    seq: 0
    stamp:
      secs: 0
      nsecs: 0
    frame_id: ''
  joint_names:
  - ''
  points:
  - num_groups: 0
    groups:
    - group_number: 0
      num_joints: 0
      valid_fields: 0
      positions: [0.2, 0, 0, 0, 0, 0]
      velocities: [0.1, 0, 0, 0, 0, 0]
      accelerations: [0]
      effort: [0]
      time_from_start: {secs: 2, nsecs: 0}" 
code: 
  val: 1
```

gavanderhoorn, 2021-12-23:
Please run the following command instead of what you've shown:

```
roslaunch --screen motoman_driver miso_yaskawa.launch
```

I don't believe you did that, as I would expect different output in that case.

And there's no use in trying anything with the driver in this state, as it doesn't initialise correctly.

I'm also unsure why you'd want to use the `joint_path_command` service. The preference would be to use the `FollowJointTrajectoryAction` server.

PS: are you running everything as `root`?

PPS: the `rosservice` invocation you show would most likely not result in (correct) motion. All list fields for each traj pt should be of equal length, but ignoring that: you've set `valid_fields` to `0`, which means: no fields contain valid information.

But again: please only use the `FollowJointTrajectoryAction` server for now. If you're just starting out, any other interface will only confuse.

rsrisamang-miso, 2021-12-23:
Sorry for misunderstanding your --screen option!  Here's the output for the following command:

I am running everything as root in my shell.  Also, I updated the joint_names_gp7.yaml file to include the additional external motor:

File: 
```
controller_joint_names: ['joint_1_s', 'joint_2_l', 'joint_3_u', 'joint_4_r', 'joint_5_b', 'joint_6_t', 'joint_1_rail']
```

Command:
```
roslaunch --screen motoman_driver miso_yaskawa.launch
```

Output:
```
roslaunch --screen motoman_driver miso_yaskawa.launch 
... logging to /root/.ros/log/301acf0c-63f1-11ec-a62d-0242ac120005/roslaunch-flippy-237142.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://flippy:40381/

SUMMARY
========

PARAMETERS
 * /controller_joint_names: ['joint_1_s', 'jo...
 * /robot_ip_address: 10.191.0.40
 * /rosdistro: kinetic
 * /rosversion: 10.2.0
 * /topic_list: [{'joints': ['joi...

NODES
  /
    io_relay (motoman_driver/io_relay)
    joint_state (motoman_driver/robot_state)
    joint_trajectory_action (industrial_robot_client/joint_trajectory_action)
    motion_streaming_interface (motoman_driver/motion_streaming_interface)

auto-starting new master
process[master]: started with pid [237190]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 301acf0c-63f1-11ec-a62d-0242ac120005
process[rosout-1]: started with pid [237222]
started core service [/rosout]
process[joint_state-2]: started with pid [237233]
[ INFO] [1640264809.819306644 /joint_state] [/root/workspace/src/industrial_core/simple_message/src/message_manager.cpp:message_manager::MessageManager::add:217]: Added message handler for message type: 0
[ INFO] [1640264809.820033882 /joint_state] [/root/workspace/src/motoman/motoman_driver/src/industrial_robot_client/robot_state_interface.cpp:robot_state_interface::RobotStateInterface::init:78]: Robot state connecting to IP address: '10.191.0.40:50241'
[ INFO] [1640264809.820660901 /joint_state] [/root/workspace/src/motoman/motoman_driver/src/industrial_robot_client/motoman_utils.cpp:motoman_utils::getJointGroups:55]: Loading topic list
[ INFO] [1640264809.820674439 /joint_state] [/root/workspace/src/motoman/motoman_driver/src/industrial_robot_client/motoman_utils.cpp:motoman_utils::getJointGroups:56]: Found 2 topics
[ INFO] [1640264809.820684861 /joint_state] [/root/workspace/src/motoman/motoman_driver/src/industrial_robot_client/motoman_utils.cpp:motoman_utils::getJointGroups:62]: Topic(state_value): [group:0,joints:{joint_1_s,joint_2_l,joint_3_u,joint_4_r,joint_5_b,joint_6_t},name:gp7_yrc1000u,ns:gp7]
[ INFO] [1640264809.820731522 /joint_state] [/root/workspace/src/motoman/motoman_driver/src/industrial_robot_client/motoman_utils.cpp:motoman_utils::getJointGroups:62]: Topic(state_value): [group:1,joints:{joint_1_rail},name:gp7_yrc1000u_rail,ns:gp7]
[ INFO] [1640264809.820750063 /joint_state] [/root/workspace/src/motoman/motoman_driver/src/industrial_robot_client/motoman_utils.cpp:motoman_utils::getJointGroups:69]: Loading group: [group:0,joints:{joint_1_s,joint_2_l,joint_3_u,joint_4_r,joint_5_b,joint_6_t},name:gp7_yrc1000u,ns:gp7]
[ INFO] [1640264809.820846987 /joint_state] [/root/workspace/src/motoman/motoman_driver/src/industrial_robot_client/motoman_utils.cpp:motoman_utils::getJointGroups:69]: Loading group: [group:1,joints:{joint_1_rail},name:gp7_yrc1000u_rail,ns:gp7]
[ INFO] [1640264809.820875982 /joint_state] [/root/workspace/src/motoman/motoman_driver/src/industrial_robot_client/motoman_utils.cpp:motoman_utils::getJointGroups:114]: Loaded 2 groups
[ INFO] [1640264809.820911465 /joint_state] [/root/workspace/src/motoman/motoman_driver/src/industrial_robot_client/robot_state_interface.cpp:robot_state_interface::RobotStateInterface::init:108]:  Initializing robot state 2 groups
[ INFO] [1640264809.820925035 /joint_state] [/root/workspace/src/industrial_core/simple_message/src/message_manager.cpp:message_manager::MessageManager::init:80]: Initializing message manager with default comms fault handler
[ INFO] [1640264809.820932378 /joint_state] [/root/workspace/src/industrial_core/simple_message/src/simple_comms_fault_handler.cpp:simple_comms_fault_handler::SimpleCommsFaultHandler::init:62]: Default communications fault handler successfully initialized
[ INFO] [1640264809.820941157 /joint_state] [/root/workspace/src/industrial_core/simple_message/src/message_manager.cpp:message_manager::MessageManager::init:102]: Initializing message manager
[ INFO] [1640264809.820950363 /joint_state] [/root/workspace/src/industrial_core/simple_message/src/message_manager.cpp:message_manager::MessageManager::add:217]: Added message handler for message type: 1
[ INFO] [1640264809.822514275 /joint_state] [/root/workspace/src/industrial_core/simple_message/src/message_manager.cpp:message_manager::MessageManager::add:217]: Added message handler for message type: 10
[ INFO] [1640264809.822890348 /joint_state] [/root/workspace/src/industrial_core/simple_message/src/message_manager.cpp:message_manager::MessageManager::add:217]: Added message handler for message type: 15
[ INFO] [1640264809.822907812 /joint_state] [/root/workspace/src/motoman/motoman_driver/src/industrial_robot_client/joint_feedback_ex_relay_handler.cpp:joint_feedback_ex_relay_handler::JointFeedbackExRelayHandler::init:54]: Creating joint_feedback_ex_relay_handler with 2 groups
[... cut ...]
/gp7/gp7_yrc1000u/stop_motion
/gp7/gp7_yrc1000u_rail/joint_path_command
/gp7/gp7_yrc1000u_rail/stop_motion
/io_relay/get_loggers
/io_relay/set_logger_level
/joint_path_command
/joint_state/get_loggers
/joint_state/set_logger_level
/joint_trajectory_action/get_loggers
/joint_trajectory_action/set_logger_level
/motion_streaming_interface/get_loggers
/motion_streaming_interface/set_logger_level
/read_single_io
/robot_disable
/robot_enable
/rosout/get_loggers
/rosout/set_logger_level
/stop_motion
/write_single_io
```

rsrisamang-miso, 2021-12-23:
I think I found the issue.  The version0 value is set to 'true' in the launch file, which assumes older client version.  I set it to false and now I don't get the error.

gavanderhoorn, 2021-12-23:
Ok. It seems you're using an incorrect `.launch` file.

You should be `include`-ing the `$(find motoman_driver)/launch/robot_multigroup_interface_streaming_$(arg controller).launch` file.

rsrisamang-miso, 2021-12-23:
Thank you for your help @gavanderhoorn, and @akashjinandra too!  Definitely learned something here.  I appreciate your advice and comments on this issue.

gavanderhoorn, 2021-12-23:
As it appears we've diagnosed the cause of the error reported in the OP, I'm going to close this issue.

closed 2021-12-23 by gavanderhoorn