ros-industrial/motoman #446: git fetch origin pull/215/head:point_streaming

MaxorPaxor, 2022-01-19:
# Summary
Hello,
I'm trying to implement point_streaming

# Environment
OS : Ubuntu 20.04 (Tried on 18.04 as well)
ROS Version : Noetic (Tried on Melodic as well)
Robot : GP8
Controller : ycr1000

# Details
After cloning and building this repo, I try to merge #215 by:
`git fetch origin pull/215/head:point_streaming`
`git checkout point_streaming`
`catkin build`

And it fails while building it with the next errors:

<details>
<summary>Full Errors</summary>
<br>
Starting  >>> motoman_driver                                                                               
___________________________________________________________________________________________________________
Errors     << motoman_driver:make /home/roblab4/max_ws/logs/motoman_driver/build.make.007.log              
In file included from /opt/ros/noetic/include/simple_message/socket/tcp_socket.h:36,
                 from /opt/ros/noetic/include/simple_message/socket/tcp_client.h:37,
                 from /home/roblab4/max_ws/src/motoman/motoman_driver/include/motoman_driver/industrial_robot_client/robot_state_interface.h:42,
                 from /home/roblab4/max_ws/src/motoman/motoman_driver/src/industrial_robot_client/robot_state_interface.cpp:32:
/opt/ros/noetic/include/simple_message/socket/simple_socket.h:204:3: error: ‘sockaddr_in’ does not name a type
  204 |   sockaddr_in sockaddr_;
      |   ^~~~~~~~~~~
/opt/ros/noetic/include/simple_message/socket/simple_socket.h: In constructor ‘industrial::simple_socket::SimpleSocket::SimpleSocket()’:
/opt/ros/noetic/include/simple_message/socket/simple_socket.h:158:19: error: ‘class industrial::simple_socket::SimpleSocket’ has no member named ‘sockaddr_’
  158 |     memset(&this->sockaddr_, 0, sizeof(this->sockaddr_));
      |                   ^~~~~~~~~
/opt/ros/noetic/include/simple_message/socket/simple_socket.h:158:46: error: ‘class industrial::simple_socket::SimpleSocket’ has no member named ‘sockaddr_’
  158 |     memset(&this->sockaddr_, 0, sizeof(this->sockaddr_));
      |                                              ^~~~~~~~~
In file included from /opt/ros/noetic/include/simple_message/socket/tcp_socket.h:36,
                 from /opt/ros/noetic/include/simple_message/socket/tcp_client.h:37,
                 from /home/roblab4/max_ws/src/motoman/motoman_driver/include/motoman_driver/industrial_robot_client/robot_state_interface.h:42,
                 from /home/roblab4/max_ws/src/motoman/motoman_driver/src/industrial_robot_client/robot_state_interface.cpp:32:
/opt/ros/noetic/include/simple_message/socket/simple_socket.h:204:3: error: ‘sockaddr_in’ does not name a type
  204 |   sockaddr_in sockaddr_;
      |   ^~~~~~~~~~~
/opt/ros/noetic/include/simple_message/socket/simple_socket.h: In constructor ‘industrial::simple_socket::SimpleSocket::SimpleSocket()’:
/opt/ros/noetic/include/simple_message/socket/simple_socket.h:158:19: error: ‘class industrial::simple_socket::SimpleSocket’ has no member named ‘sockaddr_’
  158 |     memset(&this->sockaddr_, 0, sizeof(this->sockaddr_));
      |                   ^~~~~~~~~
/opt/ros/noetic/include/simple_message/socket/simple_socket.h:158:46: error: ‘class industrial::simple_socket::SimpleSocket’ has no member named ‘sockaddr_’
  158 |     memset(&this->sockaddr_, 0, sizeof(this->sockaddr_));
      |                                              ^~~~~~~~~
In file included from /opt/ros/noetic/include/simple_message/socket/tcp_socket.h:36,
                 from /opt/ros/noetic/include/simple_message/socket/tcp_client.h:37,
                 from /home/roblab4/max_ws/src/motoman/motoman_driver/include/motoman_driver/industrial_robot_client/joint_trajectory_interface.h:46,
                 from /home/roblab4/max_ws/src/motoman/motoman_driver/src/industrial_robot_client/joint_trajectory_interface.cpp:33:
/opt/ros/noetic/include/simple_message/socket/simple_socket.h:204:3: error: ‘sockaddr_in’ does not name a type
  204 |   sockaddr_in sockaddr_;
      |   ^~~~~~~~~~~
/opt/ros/noetic/include/simple_message/socket/simple_socket.h: In constructor ‘industrial::simple_socket::SimpleSocket::SimpleSocket()’:
/opt/ros/noetic/include/simple_message/socket/simple_socket.h:158:19: error: ‘class industrial::simple_socket::SimpleSocket’ has no member named ‘sockaddr_’
  158 |     memset(&this->sockaddr_, 0, sizeof(this->sockaddr_));
      |                   ^~~~~~~~~
/opt/ros/noetic/include/simple_message/socket/simple_socket.h:158:46: error: ‘class industrial::simple_socket::SimpleSocket’ has no member named ‘sockaddr_’
  158 |     memset(&this->sockaddr_, 0, sizeof(this->sockaddr_));
      |                                              ^~~~~~~~~
In file included from /opt/ros/noetic/include/simple_message/socket/tcp_socket.h:36,
                 from /opt/ros/noetic/include/simple_message/socket/tcp_client.h:37,
                 from /home/roblab4/max_ws/src/motoman/motoman_driver/include/motoman_driver/industrial_robot_client/joint_trajectory_interface.h:46,
                 from /home/roblab4/max_ws/src/motoman/motoman_driver/src/industrial_robot_client/joint_trajectory_interface.cpp:33:
[... 36 lines cut ...]
      |                   ^~~~~~~~~
/opt/ros/noetic/include/simple_message/socket/simple_socket.h:158:46: error: ‘class industrial::simple_socket::SimpleSocket’ has no member named ‘sockaddr_’
  158 |     memset(&this->sockaddr_, 0, sizeof(this->sockaddr_));
      |                                              ^~~~~~~~~
make[2]: *** [CMakeFiles/motoman_industrial_robot_client_bswap.dir/build.make:141: CMakeFiles/motoman_industrial_robot_client_bswap.dir/src/industrial_robot_client/robot_state_interface.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/motoman_industrial_robot_client.dir/build.make:141: CMakeFiles/motoman_industrial_robot_client.dir/src/industrial_robot_client/robot_state_interface.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/motoman_industrial_robot_client.dir/build.make:115: CMakeFiles/motoman_industrial_robot_client.dir/src/industrial_robot_client/joint_trajectory_streamer.cpp.o] Error 1
make[2]: *** [CMakeFiles/motoman_industrial_robot_client_bswap.dir/build.make:115: CMakeFiles/motoman_industrial_robot_client_bswap.dir/src/industrial_robot_client/joint_trajectory_streamer.cpp.o] Error 1
make[2]: *** [CMakeFiles/motoman_industrial_robot_client_bswap.dir/build.make:102: CMakeFiles/motoman_industrial_robot_client_bswap.dir/src/industrial_robot_client/joint_trajectory_interface.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1909: CMakeFiles/motoman_industrial_robot_client_bswap.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/motoman_industrial_robot_client.dir/build.make:102: CMakeFiles/motoman_industrial_robot_client.dir/src/industrial_robot_client/joint_trajectory_interface.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:526: CMakeFiles/motoman_industrial_robot_client.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/roblab4/max_ws/build/motoman_driver; catkin build --get-env motoman_driver | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

...........................................................................................................
Failed     << motoman_driver:make                         [ Exited with code 2 ]                           
Failed    <<< motoman_driver                              [ 2.1 seconds ]  
</details>

Was wondering if I'm doing anything wrong or is it just outdated?
Thank you very much

gavanderhoorn, 2022-01-19:
It's missing (at least) #427.

gavanderhoorn, 2022-01-19:
I'm going to close this, as it's not something we need to fix.

That branch/PR is kept around for archival purposes only.

closed 2022-01-19 by gavanderhoorn