ros-industrial/motoman #450: Joint Trajectory Action status is always ends in "LOST" after executing command

dean4ta, 2022-02-01:
## Context

When sending trajectories to a Yaskawa GP7 using an action server hosted by the [joint_trajectory_action.cpp](https://github.com/ros-industrial/motoman/blob/kinetic-devel/motoman_driver/src/industrial_robot_client/joint_trajectory_action.cpp), the status goes from ACTIVE to LOST (see [GoalStatus msg](http://docs.ros.org/en/lunar/api/actionlib_msgs/html/msg/GoalStatus.html)) regardless of the trajectory executed by the arm is a "success" (gets to its desired final position) or is interrupted by an e-stop or collision.

## Process

We bring up an Action Client to the Action Server provided by Yaskawa and send a goal that executes as expected. We read the state output and it changes from ACTIVE to LOST.

partial code from the client side:
```python

def submit_trajectory_goal(gp7_client, goal, duration_sec):
    gp7_client.send_goal(goal)
    gp7_client.wait_for_result()
    # We were logging the state and callback status here with gp7_client.get_state()
    rospy.loginfo(gp7_client.get_result())

def main():
    # create client and make sure it's available
    client = actionlib.SimpleActionClient('joint_trajectory_action',
                                           FollowJointTrajectoryAction, done)
    client.wait_for_server()

    # Create FollowJointTrajectoryGoal object
    new_joint_goal = FollowJointTrajectoryGoal()
    new_joint_goal.trajectory = JointTrajectory()
    start_time_sec = 0.0
    time_from_start_sec = 0.0
    duration_sec = 10

    # Get Current position of joints
    [gp7_initial_joint_states, joint_names_r1s1] = get_gp7_current_joint_states()
    # Initialize goal joint_names
    new_joint_goal.trajectory.joint_names = joint_names_r1s1
    # Initialize velocity
    qdot = [0.0] * len(joint_names_r1s1)
    # Initialize joint_goal to current position.
    new_joint_goal.trajectory.points.append(JointTrajectoryPoint(positions=gp7_initial_joint_states,
        velocities=qdot, time_from_start=rospy.Duration(start_time_sec)))

    # Make a copy of gp7_initial_joint_states to pass into create_joint_goal and to preserve
    # initial joint states
    new_joint_states = gp7_initial_joint_states[:]
    # Elapsed time for each joint move

    time_from_start_sec += duration_sec
    create_joint_goal(new_joint_goal, 'joint_1_s', new_joint_states, 60, time_from_start_sec)
    time_from_start_sec += duration_sec
    create_joint_home_goal(new_joint_goal, gp7_initial_joint_states, time_from_start_sec)
    rospy.loginfo("Submit joint goal : %s", new_joint_goal)
    submit_trajectory_goal(client, new_joint_goal, time_from_start_sec)
    
```

## Terminal Output

```
[INFO] [1643656915.186073 /trajectory_manager] [/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/core.py:__call__:177]: $
rajectory state: 1
[INFO] [1643656915.445529 /trajectory_manager] [/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/core.py:__call__:177]: $
rajectory state: 1
[INFO] [1643656915.683436 /trajectory_manager] [/root/workspace/src/miso-robot-driver-stack/yaskawa_robot_hw_interface/src$
yaskawa_robot_hw_interface/trajectory_manager.py:doneCb:121]: doneCb_status: 9
[INFO] [1643656915.683649 /trajectory_manager] [/root/workspace/src/miso-robot-driver-stack/yaskawa_robot_hw_interface/src$
yaskawa_robot_hw_interface/trajectory_manager.py:doneCb:122]: doneCb_result: None
[INFO] [1643656915.705677 /trajectory_manager] [/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/core.py:__call__:177]: T
rajectory state: 9
[INFO] [1643656915.965501 /trajectory_manager] [/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/core.py:__call__:177]: T
rajectory state: 9
```

it is worth noting that `Trajectory state: 1` means status is "ACTIVE" and `Trajectory state: 9` means status is "LOST". You can also see there was a `done_cb` for the action client that printed the status and result

## Expected Results

I was expecting the Motoman's ActionServer to output the status "SUCCEEDED" when the trajectory was achieved and "ABORTED" or "PREEMPTED" on collision and e-stop.

## Where I am at

I did not get very far, but I traced the launch file [robot_interface_streaming_yrc1000.launch](https://github.com/ros-industrial/motoman/blob/kinetic-devel/motoman_driver/launch/robot_interface_streaming_yrc1000.launch) to this node implementation [joint_trajectory_action.cpp](https://github.com/ros-industrial/motoman/blob/kinetic-devel/motoman_driver/src/industrial_robot_client/joint_trajectory_action.cpp) where the Action Server is hosted, but I do not see where the result or status is handled or sent.

Thanks

gavanderhoorn, 2022-02-02:
Could you please describe your setup more? Is this a single-group robot, how are you starting things, which `.launch` files are you using?

And if possible, a minimal script which reproduces what you're reporting.

References to code I cannot access complicate diagnosing what's going on.

dean4ta, 2022-02-02:
Hi thanks for the support, I will get you more concise setup information and how we are launching things later today. In the meantime, I was having difficulty finding where the Action Server actually sends results using methods like: `action_server.set_succeeded(result)`, `action_server.set_preempted()`, `action_server.set_aborted()`. Where are these resutls set?

gavanderhoorn, 2022-02-03:
`motoman_driver` is not written in Python, so I'm not surprised you couldn't find the methods you mention.

dean4ta, 2022-02-03:
## Yaskawa Action Server Launch

Okay so I gathered some data after launching the motoman drivers and Trajectory action server with the following:

top-level launch file:
`yaskawa.launch`
```xml
<launch>
    <!-- motoman driver specific -->
    <arg name="robot_ip" default="10.191.0.40" />
    <arg name="robot_namespace" default="yaskawa" />
    <arg name="jt_controller_ns" default="yaskawa" />
    <arg name="trajectory_action_name" default="follow_joint_trajectory" />
    <arg name="using_robot" default="true" />
    <!-- motoman adapter specific -->
    <arg name="read_single_io_srv_name" default="read_single_io"/>
    <arg name="gripper_state_addr" default="10"/>
    <arg name="service_call_rate" default="10"/>
    <arg name="write_single_io_srv_name" default="write_single_io"/>
    <arg name="close_gripper_address" default="10010"/>
    <arg name="open_gripper_address" default="10011"/>
    <rosparam command="load" file="$(find yaskawa_robot_hw_interface)/config/joint_names_gp7.yaml" />
    <rosparam command="load" file="$(find yaskawa_robot_hw_interface)/config/gp7_motion_interface.yaml" />

    <node name="joint_state_filter" pkg="yaskawa_robot_hw_interface" type="joint_state_filter" />

    <remap from="/joint_states" to="/joint_states_raw"/>
    <remap from="/joint_trajectory_action/cancel" to="/$(arg jt_controller_ns)/$(arg trajectory_action_name)/cancel"/>
    <remap from="/joint_trajectory_action/feedback" to="/$(arg jt_controller_ns)/$(arg trajectory_action_name)/feedback"/>
    <remap from="/joint_trajectory_action/goal" to="/$(arg jt_controller_ns)/$(arg trajectory_action_name)/goal"/>
    <remap from="/joint_trajectory_action/result" to="/$(arg jt_controller_ns)/$(arg trajectory_action_name)/result"/>
    <remap from="/joint_trajectory_action/status" to="/$(arg jt_controller_ns)/$(arg trajectory_action_name)/status"/>
    <include if="$(arg using_robot)" file="$(find motoman_driver)/launch/robot_multigroup_interface_streaming_yrc1000.launch">
        <arg name="robot_ip" value="$(arg robot_ip)" />
    </include>

    <include file="$(find yaskawa_robot_hw_interface)/launch/motoman_adapter.launch">
        <arg name="read_single_io_srv_name" value="$(arg read_single_io_srv_name)" />
        <arg name="gripper_state_addr" value="$(arg gripper_state_addr)" />
        <arg name="service_call_rate" value="$(arg service_call_rate)" />
        <arg name="write_single_io_srv_name" value="$(arg write_single_io_srv_name)" />
        <arg name="close_gripper_address" value="$(arg close_gripper_address)" />
        <arg name="open_gripper_address" value="$(arg open_gripper_address)" />
    </include>
</launch>
```

`joint_state_filter` node is meant to combine the published joint_states from the yaskawa arm and a slider rail into one message. (This step complicates the answer if we are using a single group or multigroup robot)

`joint_state_filter.py`
```python
import rospy

from sensor_msgs.msg import JointState

from message_filters import (
    ApproximateTimeSynchronizer,
    Subscriber,
)

class JointStateFilter(object):
    def __init__(self):
        self.joint_state_topics = [
            "gp7/gp7_yrc1000_r1s1/joint_states",
            "gp7/gp7_yrc1000_s1/joint_states",
        ]

        self.joint_state_pub = rospy.Publisher(
            "joint_states", JointState, queue_size=10
        )
[... 610 lines cut ...]
             miso_robot_msgs/IOState                     [8b35663f14aa0799d510e8cbdf5fa8eb]
             motoman_msgs/DynamicJointTrajectory         [81bfbf2d02070fdef3a528bd72b49257]
             motoman_msgs/DynamicJointTrajectoryFeedback [84d3bbf7103790ff0a8946017b895a1a]
             rosgraph_msgs/Log                           [acffd30cd6b6de30f120938c17c593fb]
             sensor_msgs/JointState                      [3066dcd76a6cfaef579bd0f34173e9fd]
topics:      /dynamic_feedback_states                                815 msgs    : motoman_msgs/DynamicJointTrajectoryFeedback
             /feedback_states                                       1628 msgs    : control_msgs/FollowJointTrajectoryFeedback 
             /gp7/gp7_yrc1000_r1s1/feedback_states                   816 msgs    : control_msgs/FollowJointTrajectoryFeedback 
             /gp7/gp7_yrc1000_r1s1/joint_states                      814 msgs    : sensor_msgs/JointState                     
             /gp7/gp7_yrc1000_r1s1/joint_trajectory_action/status    111 msgs    : actionlib_msgs/GoalStatusArray             
             /gp7/gp7_yrc1000_s1/feedback_states                     815 msgs    : control_msgs/FollowJointTrajectoryFeedback 
             /gp7/gp7_yrc1000_s1/joint_states                        815 msgs    : sensor_msgs/JointState                     
             /gp7/gp7_yrc1000_s1/joint_trajectory_action/status      111 msgs    : actionlib_msgs/GoalStatusArray             
             /joint_path_command                                       1 msg     : motoman_msgs/DynamicJointTrajectory        
             /joint_states                                           815 msgs    : sensor_msgs/JointState                     
             /joint_states_raw                                      1630 msgs    : sensor_msgs/JointState                     
             /motoman_adapter/gripper_state                          219 msgs    : miso_robot_msgs/IOState                    
             /robot_status                                           815 msgs    : industrial_msgs/RobotStatus                
[... cut ...]             27 msgs    : rosgraph_msgs/Log                           (6 connections)
             /rosout_agg                                              14 msgs    : rosgraph_msgs/Log                          
             /yaskawa/follow_joint_trajectory/status                 112 msgs    : actionlib_msgs/GoalStatusArray             
             /yaskawa_alarms                                         470 msgs    : miso_robot_msgs/AlarmState
```

I hope this isn't too much information. Thanks

gavanderhoorn, 2022-02-07:
thanks for all the information.

> This step complicates the answer if we are using a single group or multigroup robot

You have a multi-group system. I'm not really concerned with the ROS side. MotoROS and the configuration of the Yaskawa controller determine whether you have a multi-group system or not.

The fact you mention you have an `R1` and an `S1` already imply you have a multi-group system.

> `joint_state_filter` node is meant to combine the published joint_states from the yaskawa arm and a slider rail into one message.

I'm a little confused as to why this is necessary.

`motoman_driver` publishes combined `JointState`s for all defined groups on `/joint_states` (together with separate publishers for each configured group).

Could you summarise why you found the extra script necessary?

What about the `source_list` parameter of a regular `joint_state_publisher` instead?

> ```python
> # The group numbers below are define in gp7_motion_interface.yaml file
> # That file specifies R1+S1 as Group 0, and S1 as Group 1.
> GROUP_R1S1 = 0  # R1S1 - GP7 arm and rail motor
> GROUP_S1 = 1  # S1 - station motor (rail motor)
> ```

could you clarify how your controller is configured?

Afaik MotoROS should report the joints in R1 in a message, and the joints of your S1 in a message, and the combination of those. `motoman_driver` then converts those to separate messages on the `joint_states` topics for each respective group *and* a single, combined `JointState` message for everything together.

@ted-miller @EricMarcil: is it possible for MotoROS to do what the comment states (ie: have a joint in two groups)?

> ```python
> # Get joint states for R1S1 and S1.  Note that GROUP_R1S1 doesn't contain info on
> # S1 joint position.  Must grab that info specifically from GROUP_S1.  This is an
> # idiosyncrasy of the ROS driver for the GP7.
> ```

this comment also confuses me, but perhaps you'll clear that up when you explain the need for the custom `joint_state_filter` script.

> ```xml
> <rosparam command="load" file="$(find yaskawa_robot_hw_interface)/config/joint_names_gp7.yaml" />
> <rosparam command="load" file="$(find yaskawa_robot_hw_interface)/config/gp7_motion_interface.yaml" />
> ```

you don't need both of these. Either you have a multi-group system and use the `topic_list` (ie: "motion interface `.yaml`"), or you don't, and you use `controller_joint_names`.

> `yaskawa.launch`: lots of remapping

I would recommend to see what the behaviour is if you remove all the remaps, and use a simple action client like shown in https://github.com/ros-industrial/motoman/compare/kinetic-devel...gavanderhoorn:actionclient_example.

### actions

please:

 1. include a screenshot of `rqt_graph`. I'm wondering whether there are incorrect connections due to the remaps you've introduced
 1. show the output of `roslaunch --screen /path/to/yaskawa.launch` (or via the package it is in, doesn't matter). We don't need everything, but make sure to copy-paste all lines originating from the nodes in `motoman_driver`.
 1. show the contents of `yaskawa_robot_hw_interface/config/gp7_motion_interface.yaml`
 1. explain the need for `joint_state_filter`
 1. explain how your Yaskawa controller is setup wrt motion groups
 1. provide a whireshark trace of the traffic between `motoman_driver` and `MotoROS` (ie: the traffic between the Yaskawa controller and the ROS nodes when executing a trajectory)
 1. try again without remapping anything, and using a simple action client

ted-miller, 2022-02-07:
> > ```python
> > # The group numbers below are define in gp7_motion_interface.yaml file
> > # That file specifies R1+S1 as Group 0, and S1 as Group 1.
> > GROUP_R1S1 = 0  # R1S1 - GP7 arm and rail motor
> > GROUP_S1 = 1  # S1 - station motor (rail motor)
> > ```
> @ted-miller @EricMarcil: is it possible for MotoROS to do what the comment states (ie: have a joint in two groups)?

No, not with the ROS interface.

On the teach pendant, there are "group combinations" that can be defined. These can appear to be a single group when programming an Inform job. However, they are distinct separate groups at a lower level.

As you stated above, the MotoROS interface will send three feedback messages to `motoman_driver`. One for R1, one for S1, and one with both groups.

ted-miller, 2022-02-07:
To elaborate a little more, it can be little extra confusing when using a rail mounted to the robot's base. The rail is implicitly linked to R1 without defining a "group combination". This makes it part of the robot's kinematic chain. But, the rail is still an independent group at the lowest level.

gavanderhoorn, 2022-02-07:
Yes, that's how our YRC1000 is also configured (GP25 on a TSL-1000).

nardavin, 2022-02-08:
Hi, I'm another engineer working on the same project as @dean4ta - we were able to strip down our code to create a more concise example of the error we experienced, incorporating the advice we've received so far. Hopefully this helps explain what we are seeing and what is going wrong.

test_yaskawa.launch:
```
<launch>
    <arg name="robot_ip" default="10.191.0.42" />
    <rosparam command="load" file="$(find yaskawa_robot_hw_interface)/config/gp7_motion_interface.yaml" />

    <include file="$(find motoman_driver)/launch/robot_multigroup_interface_streaming_yrc1000.launch" >
        <arg name="robot_ip" value="$(arg robot_ip)" />
    </include>

    <node pkg="yaskawa_robot_hw_interface" type="joint_traj_client" name="joint_traj_client" output="screen" />

</launch>
```

gp7_motion_interface.yaml:
```
topic_list:
    - name: gp7_yrc1000_r1
      ns: gp7
      group: 0
      joints: ['joint_1_s', 'joint_2_l', 'joint_3_u', 'joint_4_r', 'joint_5_b', 'joint_6_t']
    - name: gp7_yrc1000_s1
      ns: gp7
      group: 1
      joints: ['joint_1_rail']
```

joint_traj_client:
```
#!/usr/bin/env python

# TODO: add license

# A simple example showing the use of an Action client to request execution of
# a complete JointTrajectory.
#
# Note: joint names must match those used in the urdf, or the driver will
#       refuse to accept the goal.

import rospy
import actionlib
import time
import math
import sys

from control_msgs.msg import FollowJointTrajectoryAction, FollowJointTrajectoryGoal
from sensor_msgs.msg import JointState
from trajectory_msgs.msg import JointTrajectory, JointTrajectoryPoint
from industrial_msgs.msg import RobotStatus

arm = False
rail = False
start_pos = [0.0] * 7

def joint_state_cb(msg):
    # Find the starting position of all joints among the two groups
    # This is just a hacky way to get the entire robot's initial joints without 
    # having to set up a joint_state_publisher, which eliminates some 
    # potential issues for this simple test
    global arm
    global rail
    global start_pos

    if not arm and 'joint_1_s' in msg.name:
        arm = True
        rospy.loginfo('Got arm message')
        for i in range(6):
[... 91 lines cut ...]
des/joint_traj_client:joint_state_cb:41]: Got rail message
[INFO] [1644281699.274389 /joint_traj_client] [/root/workspace/src/miso-robot-driver-stack/yaskawa_robot_hw_interface/no
des/joint_traj_client:joint_state_cb:36]: Got arm message
[INFO] [1644281699.324443 /joint_traj_client] [/root/workspace/src/miso-robot-driver-stack/yaskawa_robot_hw_interface/no
des/joint_traj_client:main:72]: Got full state
[INFO] [1644281699.324679 /joint_traj_client] [/root/workspace/src/miso-robot-driver-stack/yaskawa_robot_hw_interface/no
des/joint_traj_client:main:102]: Waiting for robot to be active
```

And after we send a `rosservice call robot_enable` in a separate terminal, we see the following output:
```
[INFO] [1644281704.761014 /joint_traj_client] [/root/workspace/src/miso-robot-driver-stack/yaskawa_robot_hw_interface/no
des/joint_traj_client:main:106]: Robot active!
[INFO] [1644281704.761187 /joint_traj_client] [/root/workspace/src/miso-robot-driver-stack/yaskawa_robot_hw_interface/no
des/joint_traj_client:main:110]: Submitting goal ..
[INFO] [1644281704.761398 /joint_traj_client] [/root/workspace/src/miso-robot-driver-stack/yaskawa_robot_hw_interface/no
des/joint_traj_client:main:112]: Waiting for completion ..
[INFO] [1644281710.138613 /joint_traj_client] [/root/workspace/src/miso-robot-driver-stack/yaskawa_robot_hw_interface/no
des/joint_traj_client:main:114]: Done.
[INFO] [1644281710.139182 /joint_traj_client] [/root/workspace/src/miso-robot-driver-stack/yaskawa_robot_hw_interface/no
des/joint_traj_client:main:115]: Action server state: 9
[INFO] [1644281710.139769 /joint_traj_client] [/root/workspace/src/miso-robot-driver-stack/yaskawa_robot_hw_interface/no
des/joint_traj_client:main:116]: Action server result: None
```
While waiting for completion, all joints do successfully move as we would expect. However, we get the LOST state (9) when querying the action server state, rather than SUCCEEDED (3). We also get the LOST state when terminating the trajectory early, such as through an e-stop.

gavanderhoorn, 2022-02-08:
Thanks for the update, but I'd still like to see:

 - the wireshark capture
 - the `rqt_graph` screenshot, and
 - the output of the `motoman_driver` nodes when you start everything with `roslaunch --screen ...`

> ```python
> def joint_state_cb(msg):
>     # Find the starting position of all joints among the two groups
>     # This is just a hacky way to get the entire robot's initial joints without 
>     # having to set up a joint_state_publisher, which eliminates some 
>     # potential issues for this simple test
>     global arm
>     global rail
>     global start_pos
> 
>     if not arm and 'joint_1_s' in msg.name:
>         arm = True
>         rospy.loginfo('Got arm message')
>         for i in range(6):
>             start_pos[i] = msg.position[i]
>     elif not rail and 'joint_1_rail' in msg.name:
>         rail = True
>         rospy.loginfo('Got rail message')
>         start_pos[6] = msg.position[0]
> ```

this should not be necessary, provided you subscribe to the correct topic.

On my personal setup, I use a single `rospy.wait_for_message('joint_states', JointState)` which gets me the full `JointState`, including the track.

The `/joint_states` topic should publish messages containing the full `JointState`. The `/gp7/gp7_yrc1000_r1/joint_states` topic should carry messages containing just the `JointState`s of the `gp7_yrc1000_r1` group, and the `/gp7/gp7_yrc1000_s1/joint_states` should have them for just the `gp7_yrc1000_s1` group.

Does that not work for you?

gavanderhoorn, 2022-02-08:
Seeing as you have a multi-group system, it's likely you're being bitten by what #259 attemps to fix. It's not complete (hence why it hasn't been merged), but there's a good chance it'll resolve the lost goals.

rsrisamang-miso, 2022-02-08:
Hi @gavanderhoorn , I'm not sure why, but we are not seeing the /joint_states message containing all joint states in one frame.  Would that data be originating from the MotoROS Application or is that something the motoman_driver will generate and publish to the /joint_states topic?

gavanderhoorn, 2022-02-08:
Please provide me with at least the wireshark trace, so we can rule out a couple of things.

I'd also still like the rest of the items I described in https://github.com/ros-industrial/motoman/issues/450#issuecomment-1031460588.

gavanderhoorn, 2022-02-11:
Friendly ping

closed 2022-02-14 by gavanderhoorn

rsrisamang-miso, 2022-02-14:
Hi @gavanderhoorn, sorry for the late response.  Sorry for the late response.  Will send you info in the next day or two.  Thanks for following up.

gavanderhoorn, 2022-02-22:
Just checking you've resolved the issue.

rsrisamang-miso, 2022-02-23:
Yes, we've applied similar changes to the issue you mentioned in #259.  We were able to resolve the lost goals.  Thank you.

gavanderhoorn, 2022-02-23:
> we've applied similar changes

I'd be interested to know exactly *which* changes. That could help resolve it here as well.

gavanderhoorn, 2022-03-09:
And another ping.

gavanderhoorn, 2022-03-14:
And another friendly ping.

@dean4ta @nardavin @rsrisamang-miso: could you please respond to my request?

gavanderhoorn, 2022-03-21:
And another ping.

jmarsik, 2022-03-22:
I would be also interested in this info, could you elaborate @rsrisamang-miso?
See also my comment https://github.com/ros-industrial/motoman/pull/259#issuecomment-1074784596

gavanderhoorn, 2022-03-23:
You may be interested in the combination of https://github.com/ros-industrial/motoman/compare/kinetic-devel...gavanderhoorn:coalesced_joint_feedback_ex_relay and https://github.com/ros-industrial/motoman/compare/kinetic-devel...gavanderhoorn:rebased_two_arms_on_a_rail as well @dean4ta @nardavin @rsrisamang-miso

nardavin, 2022-03-23:
Hi all, sorry for leaving everyone hanging on this. Our changes have been in a constant state of "mostly working" for the last couple weeks, but we believe that our version of the motoman driver is now stable and correctly supports multigroup motion, at least for our hardware configuration. I'll make a PR in the next couple days once we have final approval on our end.

gavanderhoorn, 2022-03-24:
A pointer to a branch should be enough.

gavanderhoorn, 2022-03-29:
Friendly ping @nardavin

nardavin, 2022-03-29:
https://github.com/ros-industrial/motoman/pull/488