ros-industrial/motoman #451: Aborting goal because we haven't heard from the controller in 1 seconds

akashjinandra, 2022-02-01:
Hello I am currently experiencing this error: 
[image]

I am on running Ubuntu 18.04 on ROS Melodic. Currently on this branch: https://github.com/ros-industrial/motoman/pull/259

Interestingly enough I only have this error when I have an IDE open, like Qt creator or Pycharm. System specs: 
[image]
With i7 7700k

gavanderhoorn, 2022-02-02:
Please see https://github.com/ros-industrial/motoman/issues/445#issuecomment-1027730682 about configuring your console logging such that we can see what is printing those lines.

Please also start your `.launch` file with `roslaunch --screen ...`, as we're missing a lot of output which can help.

akashjinandra, 2022-02-02:
Yes this is a VM. I will gather the roslaunch with --screen.

gavanderhoorn, 2022-02-09:
Any update here @akashjinandra ?