ros-industrial/motoman #454: Need to use multiple branches for application

akashjinandra, 2022-02-08:
Hello All,

I am using Ubuntu 18.04 with ROS Melodic. I have an application where I need to both use this branch: https://github.com/ros-industrial/motoman/pull/259 due to the robot having multiple groups(robot+track+articulated gripper). However I also need to use this branch as well to write registers: https://github.com/ros-industrial/motoman/pull/439. 

The current solutions is basically to have 2 different catkin work-spaces, one where I can utilize the new motoman_msgs, and the other with the fixed joint_trajectory_action nodes in order to correctly control multiple groups. 

Is there a standard way to go about solving this besides trying to merge the 2 branches?

Thanks for your thoughts.

gavanderhoorn, 2022-02-08:
You'd create a branch in your local clone into which you `git merge` both branches.

gavanderhoorn, 2022-02-08:
As this is not an issue with the packages in this repository, I'm closing it.

Feel free to keep commenting on it.

closed 2022-02-08 by gavanderhoorn