ros-industrial/motoman #455: Activation of weld guns from MotoROS

Noraboke, 2022-02-09:
Hi,
We have been using MotoROS with our GP25-12 successfully for general movement control, but would also like to move on to welding. Is there any proper way to achieve this through MotoROS? 

So far, we have been looking into writing various user output groups on/off, however, it seems like the only relevant addresses signalize that we are ready for the welding and not actual activation of the ARCON function.

**Controller:** YRC1000
**Robot:** GP25-12
**Welding equipment:** Fronius

gavanderhoorn, 2022-02-09:
Could you clarify whether you're also using `motoman_driver`?

From the way you phrase your question I have the feeling you're not.

Noraboke, 2022-02-09:
My bad. Yes, we are also using the motoman_driver.

ted-miller, 2022-02-09:
Hi Noraboke.
Unfortunately, the ROS interface does not have any direct connection to the robot's standard "process" controls (welding, painting, cutting, etc). So, there is no way to directly call a normal ARCON/ARCOF command. But there are a couple work arounds.
(FWIW, this is on the development planning roadmap for sometime in the future. But, there's no schedule for it yet.)

Options:
1) The best option is to directly control the welder from your application on the PC. I'm not particularly familiar with the Fronius communication protocol, but I know that the Yaskawa interface is based on Modbus TCP. I'm fairly sure that Fronius supports other communication options as well, depending on which features you need from the welder. For instance, if you're doing a basic job-mode routine where the weld settings are pre-programmed on the power source, I think there's a simple I/O interface for basic on/off and job selection.

I know there are users who have had success controlling welders directly from their ROS application. Though, there applications are proprietary and closed-source. 

If your choice of power source is flexible, I know for certain that Miller, SKS, and OTC have basic I/O interfaces using Ethernet/IP. There are [open source implementations of E/IP](https://github.com/nimbuscontrols/EIPScanner) available.  

2) I've never actually attempted this... but I think you could play some games with the INIT_ROS job and control the execution flow with I/O signals (which are supported in motoman_driver). (I've been curious to see someone try this.)

<details><summary>Example modified INIT_ROS job:</summary>
<p>

```
NOP
'INIT SIGNALS
'INTERNAL MOTION SIGNALS
DOUT OT#(890) OFF
DOUT OT#(889) OFF
' 
*LOOP
TIMER T=0.05
' 
JUMP *STARTARC IF IN#(1)=ON
JUMP *ENDARC IF IN#(2)=ON
JUMP *MOTION IF IN#(3)=ON
JUMP *LOOP
' 
*STARTARC
ARCON
WAIT IN#(1)=OFF
JUMP *LOOP
' 
*ENDARC
ARCOF
WAIT IN#(2)=OFF
JUMP *LOOP
' 
*MOTION
'READY FOR MOTION
DOUT OT#(889) ON
'MOV UNTIL 890=ON
WAIT OT#(890)=ON
DOUT OT#(890) OFF
WAIT IN#(3)=OFF
JUMP *LOOP
END
```

</p>
</details>

In this example, you use OT 1-3 to determine which 'action' the robot should perform. 
- Turn on OT#3 to enable robot motion. 
- Send a trajectory to move the robot to the start of the weld seam. 
- Turn off OT 890 and OT 3 to disable motion-mode.
- Turn on OT 1 to invoke the ARCON command.
- Turn off OT 1, turn on OT 3 to enable motion again.
- Send a trajectory for motion while the welder is on.

3) Lastly, you could develop an interface to generate INFORM jobs instead of using motoman_driver. After you generate a trajectory and build an INFORM job, you can send the job to the robot using Yaskawa's High Speed Ethernet Server protocol.
I imagine this would be a lot of work.