ros-industrial/motoman #471: the following packages/stacks could not have their rosdep keys resolved to system dependencies

josephbhlee, 2022-02-23:
When I execute rosdep install command, I faced the following error message.
Can you guide me to solve this problem?

$ rosdep install --from-paths src/ --ignore-src --rosdistro kinetic
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
motoman_ms210_moveit_config: Cannot locate rosdep definition for [trac_ik_kinematics_plugin]

gavanderhoorn, 2022-02-23:
First: please note that ROS Kinetic has been EOL for about a year now. The branch here may be called `kinetic-devel`, but that's just because we haven't had a reason to create a `melodic-devel` (or `noetic-devel`) yet.

Having written that: `trac_ik_kinematics_plugin` should be available for ROS Kinetic. See [status_page/ros_kinetic_default.html?q=trac_ik](http://repositories.ros.org/status_page/ros_kinetic_default.html?q=trac_ik) for instance.

Please make sure you've configured the ROS repositories correctly, and that you're actually using `python-rosdep` from the ROS repositories, and not the `rosdep` package provided by Ubuntu directly (see [wiki/UpstreamPackages](http://wiki.ros.org/UpstreamPackages)).

gavanderhoorn, 2022-02-23:
As I don't believe this is a problem with the packages in this repository, I'm going to close the issue.

If it turns out we do need to fix something, we can re-open.

Feel free to keep commenting on it of course.

closed 2022-02-23 by gavanderhoorn

josephbhlee, 2022-02-23:
Thank you! I solved the issue by your recommendation!

gavanderhoorn, 2022-02-23:
Which one?

josephbhlee, 2022-02-23:
Having written that: trac_ik_kinematics_plugin should be available for ROS Kinetic. See [status_page/ros_kinetic_default.html?q=trac_ik](http://repositories.ros.org/status_page/ros_kinetic_default.html?q=trac_ik) for instance.

gavanderhoorn, 2022-02-23:
I'm not sure I understand.

Could you describe what you've done exactly?

josephbhlee, 2022-02-23:
I just installed the package you mentioned as follows.

sudo apt-get install ros-kinetic-trac-ik-kinematics-plugin

Thanks!