ros-industrial/motoman #490: [ MotomanGP50 + YRC1000 ] [ MoveIt ] ERROR: Parameter [/controller_list] is not set // Returned 0 controllers in list

Ivan-AITIIP, 2022-03-30:
Hi!
I am having a configuration problem with a Motoman G50 robot while configuring the conexion with ROS through MoveIt.

# Summary

- I can read the robot state & view it in Rviz
- MoveIt commands from Rviz yield error:
- Unable to identify any set of controllers that can actuate the specified joints, ABORTED: Solution found but controller failed during execution.
- I already called rosservice call /robot_enable and calling it again does not change the behavior
- Why can't I move the robot?

What test can I run to verify configuration?
How can I set the controller right?

# Hardware

- Yaskawa Motoman GP50
- YRC1000 controller
- ROS Noetic
- Ubuntu 20.04 (LTS) 

# Problem

I have managed to read the robot's topics both from RVIZ with

```
roslaunch motoman_gp50_support robot_state_visualize_gp50.launch controller:=yrc1000 robot_ip:=192.168.255.1
```

As from Moveit (with RVIZ) with

```
roslaunch motoman_gp50_moveit_config moveit_planning_execution.launch sim:=false robot_ip:=192.168.255.1 controller:=yrc1000
```

It allows me to enable the motors:
```
rosservice call /robot_enable
success: True
message: "Motoman robot is now enabled and will accept motion commands."
```

After executing this command, the robot's INIT_ROS job is activated:
[image]
and it goes to the line 5th waiting for a instruction
[image]

But when I try to move the robot (with _plan and execute_ in RViz), I get the following error:

```
[ INFO] [1648564352.733447861]: Returned 0 controllers in list
[ERROR] [1648564352.733472306]: Unable to identify any set of controllers that can actuate the specified joints: [ joint_1_s joint_2_l joint_3_u joint_4_r joint_5_b joint_6_t ]
[ERROR] [1648564352.733485179]: Known controllers and their joints:

[ERROR] [1648564352.733508309]: Apparently trajectory initialization failed
[ INFO] [1648564352.745265359]: ABORTED: Solution found but controller failed during execution
```

If I ask for the controller_list:
```
rosparam get /controller_list
ERROR: Parameter [/controller_list] is not set
```

On the robot I have the key in remote. 

Any idea what could be happening?

Here is the motoman_gp50_moveit_config folder:
[motoman_gp50_moveit_config.zip](https://github.com/ros-industrial/motoman/files/8380711/motoman_gp50_moveit_config.zip)
 
Thanks in advance for any help!

gavanderhoorn, 2022-03-30:
From the console output you show it would appear your MoveIt configuration is not complete. That's why MoveIt is complaining.

The "controllers" mentioned in the MoveIt error message refers to a MoveIt "controller", not `motoman_driver` nor anything related to your YRC1000.

`motoman_driver` seems to function correctly.

I don't have time to debug your MoveIt configuration right now, but I'd suggest checking the output of MoveIt as it starts up (ie: when you `roslaunch motoman_gp50_moveit_config moveit_planning_execution.launch [...]`). At some point MoveIt should show you something like the following:

```
...
[ INFO] [1648650625.753948060] [/move_group]: Added FollowJointTrajectory controller for
[ INFO] [1648650625.754062260] [/move_group]: Returned 1 controllers in list
[ INFO] [1648650625.764073239] [/move_group]: Trajectory execution is managing controllers
...
```

If you don't see this, MoveIt will print the error message you quoted (ie: "can't find controllers for joints ...").

marip8, 2022-03-30:
Looking through your launch files, I think this is a somewhat arcane launch file parameterization issue. In your `moveit_planning_execution.launch` file you launch the `move_group.launch` file with no argument for the `moveit_controller_manager`, so the `move_group.launch` provides a default value of `simple`, which causes `simple_moveit_controller_manager.launch.xml` to be launched from `trajectory_execution.launch.xml`. What you really want is for `motoman_gp50_moveit_controller_manager.launch.xml` to be launched instead because it loads the right parameters from your `controllers.yaml` file.

If I'm right about this, the solution should be:

1. Rename `moveit_gp50_moveit_controller_manager.launch` to `moveit_gp50_moveit_controller_manager.launch**.xml**`
    - `trajectory_execution.launch.xml` prefixes the hard-coded name `_moveit_controller_manager.launch.xml` with the `moveit_controller_manager` parameter value (i.e. `moveit_gp50`), so if that launch file doesn't have the right extension, you'll get another error
2. Update your `moveit_planning_execution.launch` file to specify the correct `moveit_controller_manager` parameter
    ```diff
    <include file="$(find motoman_gp50_moveit_config)/launch/move_group.launch">
      <arg name="publish_monitored_planning_scene" value="true" />
    +  <arg name="moveit_controller_manager" value="motoman_gp50" />
    </include>
    ```

marip8, 2022-03-30:
To solve this particular issue, both 1 and 2 will need to be done (given the linked MoveIt config package). I thought the MSA auto-generates the `<your_package_name>_moveit_controller_manager.launch.xml` file, so I'm surprised this one was missing the `.xml` extension (unless it was accidentally changed by @Ivan-AITIIP). 

Going forward, I think we should update our tutorials (the one linked by @gavanderhoorn and the one on the [training website](https://industrial-training-master.readthedocs.io/en/melodic/_source/session3/Build-a-Moveit!-Package.html)) to add the `moveit_controller_manager` parameter in `moveit_planning_execution.launch` if the MSA defaults to the value `simple` instead of the name of the robot

Ivan-AITIIP, 2022-03-31:
Thank you so much!
The robot is moving now!

Writing this in `trajectory_execution.launch.xml`
```XML
<!-- Load the robot specific controller manager; this sets the moveit_controller_manager ROS parameter -->
  <arg name="moveit_controller_manager" default="motoman_gp50" />
  <include file="$(find motoman_gp50_moveit_config)/launch/$(arg moveit_controller_manager)_moveit_controller_manager.launch.xml" />
```

and update the 'moveit_planning_execution.launch' as you tell everything works!

About the problem with the extension, yes, it was my bad. I accidentally did it.

gavanderhoorn, 2022-03-31:
Ok, good to hear you've got your MoveIt config sorted out.

closed 2022-03-31 by gavanderhoorn