ros-industrial/motoman #498: DX200: MotoRos not started (5008) and "Subtask: Neustart nicht möglich" (restart not possible)

cjue, 2022-05-05:
We installed and configured MotoRos 1.9.10 according to the current installation tutorial (including `Cycle Switch in Cycle mode`) on a DX200 controller connected to a MH100 manipulator.

Unfortunately MotoRos does not work as expected, we get the expected joint states but /robot_enable does not work:

```
Failed to set TrajectoryMode: Not Ready (5) : MotoRos not started (5008)
```

On the Standard Pendant the following message popped up after every enable attempt:
```
Subtask: Neustart nicht möglich
```
Which should translate to 
```
Subtask: Restart not possible
```

Did anyone encounter a similar error before?

Tutorial used:
http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer#Controller_Configuration

cc @larslevity

EricMarcil, 2022-05-05:
@cjue You can refer to the trouble shooting section.  
[http://wiki.ros.org/motoman_driver/Troubleshooting#A.285008.29_MotoRos_not_started](http://wiki.ros.org/motoman_driver/Troubleshooting#A.285008.29_MotoRos_not_startedl)
Most probably is that you can't playback a job manually either putting the pendant switch to PLAY mode, pressing "Servo On Ready" and then the green start button.  You'll get a message on the pendant saying why.  You probably need to do a "Check Second Home".  You can refer to section 8.2.2 of the YRC1000 instruction manual 178642-1CD or search for "Second Home Position" in your controller manual, if you have a different one.

larslevity, 2022-05-12:
Thanks for the answer!
We followed your tips, but unfortunately the problem persists.

No matter which job we select and then switch to play mode, we don't get any error message. The play mode does not become active either.

We have executed the approach of a second home position. There was no deviation here.

We also tried to solve the problem by deleting all existing jobs on the controller, but this was also unsuccessful.

As soon as the remote mode is activated, the error message "Subtask - Neustart nicht möglich" appears on the standard pendant and the error message "5008" appears on ROS side. 

Do you have any other idea where this error could come from?

EricMarcil, 2022-05-12:
If you can't start a job in PLAY mode by turning on the SERVO and then pressing the START button, then the problem is not related to ROS.  There should be a message when you press the button and nothing happens.  Try pressing the message area to expand it, there might be more messages.  I would suggest contacting your local Yaskawa/Motoman tech support to resolve the playback issue.

cjue, 2022-05-24:
> If you can't start a job in PLAY mode by turning on the SERVO and then pressing the START button, then the problem is not related to ROS. 

Thank you for your feedback, it appears to be a controller misconfiguration issue as no Jobs can be run in PLAY mode.

I'll close this for now as as we wait for the local Yaskawa tech support.

closed 2022-05-24 by cjue