ros-industrial/motoman #519: Missing `else` condition

ted-miller, 2022-08-29:
A user has found a missing `else` condition. This leads to an improper response to an unknown message type.

https://github.com/ros-industrial/motoman/blob/7860ff545106d98ed6a3fa0121f2fb60891f69bd/motoman_driver/MotoPlus/SimpleMessage.c#L153

gavanderhoorn, 2022-08-30:
Isn't the actual consequence the `else` will always be taken if `receiveMsg->header.msgType != ROS_MSG_MOTO_SELECT_TOOL`?

So https://github.com/ros-industrial/motoman/blob/7860ff545106d98ed6a3fa0121f2fb60891f69bd/motoman_driver/MotoPlus/SimpleMessage.c#L160-L162

will always be executed?

EricMarcil, 2022-08-30:
Without the "else" on line 153, all the preceeding code (line 138 to 152) is pointless because whatever is set in that code gets overwritten by line 161 and 162.

gavanderhoorn, 2022-08-30:
Yes. That is what I wrote.

ted-miller, 2022-08-30:
> Isn't the actual consequence the else will always be taken if receiveMsg->header.msgType != ROS_MSG_MOTO_SELECT_TOOL?

Yes, you're correct. Now that I think about it some more... this might explain some wireshark captures that I've seen in the past that had unexpected sequence-id's in the reply. (I don't remember for certain. But I think I've seen this and just ignored it.)

gavanderhoorn, 2023-06-28:
Closing as #550 was merged (replacement PR).

closed 2023-06-28 by gavanderhoorn