ros-industrial/motoman #524: Tight loop causing delays in processing

ted-miller, 2022-10-28:
This loop has no sleep commands
https://github.com/ros-industrial/motoman/blob/7860ff545106d98ed6a3fa0121f2fb60891f69bd/motoman_driver/MotoPlus/MotionServer.c#L1434

A user has reported that with three groups, one (or more) of the tasks can get locked out for 0.4 - 3.0 seconds. (Using mpStopWatch inside `Ros_MotionServer_AddToIncQueueProcess`)

I have advised him to try doing a sleep every 10 or 20 iterations of the loop. If he reports that this helps, I'll open a PR.

ted-miller, 2022-11-01:
User reports that he added the sleeps in the loop and also increased the size of the internal increment queue. This appears to have resolved the issue.

ted-miller, 2023-02-02:
Hmmm, found another possible tight-loop.
https://github.com/ros-industrial/motoman/blob/e074519f7610621877e13d74a12877a41289fad0/motoman_driver/MotoPlus/Controller.c#L354

I'm not sure if `mpSelect` relinquishes the CPU when there's no timeout value. But I'm thinking a delay in this loop would be good.

gavanderhoorn, 2023-02-02:
I would be surprised if VxWorks's `select(..)` does not simply suspend the process/thread calling it if there are no file descriptors ready.

It would be a rather poor implementation of `select(..)` if it used a busy loop. Having the scheduler suspend the task -- while keeping an eye on the FDs -- is what makes `select(..)` useful.

ted-miller, 2023-02-02:
But we're specifying NULL for the timeout to get an immediate return. That's why I'm concerned that the task might not be getting suspended.

gavanderhoorn, 2023-02-02:
According to POSIX, [select(2)](https://man7.org/linux/man-pages/man2/select.2.html):

> If timeout is specified as NULL, select() blocks indefinitely waiting for a file descriptor to become ready.

now the only question would be whether VxWorks's `select(..)` follows POSIX.

According to the VxWorks 6.3 headers:

```c
int select
    (
    int              width,     /* number of bits to examine from 0 */
    fd_set *         pReadFds,  /* read fds */
    fd_set *         pWriteFds, /* write fds */
    fd_set *         pExcFds,   /* exception fds */
    struct timeval * pTimeOut   /* max time to wait, NULL = forever */
    )
```

ted-miller, 2023-02-02:
Oh... I'm an idiot. I had it in my head that the logic was reversed. 
You're correct.

gavanderhoorn, 2023-06-28:
@ted-miller: would this still be something you'd want to address?

ted-miller, 2023-06-28:
Let's close this for now. A followup email from this user indicated:

> I wanted to revisit this. It seems sleeping in this loop -  [https://github.com/ros-industrial/motoman/blob/7860ff545106d98ed6a3fa0121f2fb60891f69bd/motoman_driver/MotoPlus/MotionServer.c#L1434](https://nam10.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fros-industrial%2Fmotoman%2Fblob%2F7860ff545106d98ed6a3fa0121f2fb60891f69bd%2Fmotoman_driver%2FMotoPlus%2FMotionServer.c%23L1434&data=05%7C01%7Cted.miller%40motoman.com%7C78b450b1568b420d39a708dad8a7fa4a%7C5bff86433f854a0a80f929eae318b5ce%7C0%7C0%7C638060513247921471%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=v4uloExa3KfHPTIBPiH%2FRf9VkE1ZjMwmO0jdNOey6%2Fc%3D&reserved=0)  & increasing the queue size, did not resolve the issue.

Given that there have been no other complaints (over many years), I'm not confident that this is actually an issue.

Unfortunately, communication from the user has stopped and I do not know how the issue was ultimately resolved.

closed 2023-06-28 by ted-miller