ros-industrial/motoman #537: control motoman gp7 with 3 external axis

JACKX-M, 2023-04-24:
Hi guys, I'm using ROS to control a motoan gp7 robot (already extented 3 axis while one of them is a AC SERVO) with yrc1000 controller, and I have solved the single arm control issue.
But when I want to control the external axis, it becomes a hard task for me. Since I am a freahman in this field, I have some basic problems,
1. should I change the [```INIT_ROS.JBI```](https://github.com/ros-industrial/motoman/tree/kinetic-devel/motoman_driver/Inform/DX200%2C%20FS100%2C%20YRC1000/single_arm_with_ext_axis) provided by motoman_driver by adding 3 group?
2. how to setup the moveit package and ```moveit_planning_execution.launch``` file?
3. is it necessary to modify the configs in motoman_gp7_support folder?

Thanks a lot!

ted-miller, 2023-04-24:
Hi @JACKX-M,
I'll defer to others on questions 2 and 3. But for item 1, yes you should modify the `INIT_ROS.JBI` file so that it commands all attached control-groups.

Examples of different configurations are available here: https://github.com/ros-industrial/motoman/tree/kinetic-devel/motoman_driver/Inform/DX200%2C%20FS100%2C%20YRC1000

The exact change depends on your control-group configuration. If you post the `PANELBOX.LOG` file from your robot controller, we can advise you on which job should be used.

JACKX-M, 2023-04-24:
@ted-miller Thanks for your reply! Here is my '''PANELBOX.LOG''' file: https://drive.google.com/file/d/1-5V0KsNySiPoS8YXI4Vx4dN9b5JjnpeD/view?usp=drivesdk
I upload it on google cloud because it's too long to paste here.

JACKX-M, 2023-04-24:
Sorry for that. I have changed permissions now. https://drive.google.com/file/d/1-5V0KsNySiPoS8YXI4Vx4dN9b5JjnpeD/view?usp=drivesdk