ros-industrial/motoman #539: HC-10DTP Support Files

ted-miller, 2023-04-24:
We have a user targeting the HC-10DT**P**. 

> DT vs DTP - The T flange pattern is different, **the hand guiding ring is different, the range of motion is different**, the utility connections and utilities themselves are different and the IP rating is different.
DTXP vs DTP – The T flange pattern is different, **the hand guiding ring is different, the range of motion is different**, the utility connections and utilities themselves are different. 
(however, the manipulators look the same and are both IP67)

@EricMarcil, is this something that you can update? Or should I do it?

EricMarcil, 2023-04-24:
@ted-miller I will look into doing the update.  There are a lot of variant in the HC10 series.  
It would be the 1-06VXCP10-B0* (HC10DTP) , that's the one with the white covers.  (The -A0* have blue covers.)

EricMarcil, 2023-04-25:
Create PR #541 to add 1-06VXCP10-B0* (HC10DTP) model

gavanderhoorn, 2023-04-25:
Let's keep this open until we merge #541.

closed 2024-02-23 by ted-miller